1000001101000/Debian_on_Buffalo #167: Terrastation II Pro - Debian Bullseye does not boot installer image

GeneralGresi, 2022-12-26:
When trying to use the Debian Bullseye installer images for a TS2 Pro - TS-RITGL/R5 (initrd.buffalo and uImage.buffalo.ts2pro - renamed of course), it doesn't boot if the files are put on the hard drive on an ext3 partition, nor does it boot from TFTP.
Therefore the installer can't be run.
For example atftpd from linux spits out this error:

[image]

Timeout block 14462, and repeats indefinetly.
A windows equivalent TFTP Server just displays that the device re-requested the file.

The problem does not occur when using the debian buster image. I was able to install buster without any problem, but want to use bullseye. Installing via HDD, or via atftpd works fine.

This was tried on two different but identical terrastation with exactly the same error.

1000001101000, 2022-12-27:
Good to hear from you!

My best guess is that the auto-generated installer image has grown too large for that device to handle. That particular installer is less dependent on getting updated than the armhf installer. Try this older image and see if that boots:

https://github.com/1000001101000/Debian_on_Buffalo/blob/a4c28521e6ba6f83c4550ff24016cef352414176/Bullseye/installer_images/armel_devices/initrd.buffalo

Also, TFTP tends not to work out for the installer since it needs internet access to get past the early steps. It needs to download the SSH server package and install it before it can start SSH.

GeneralGresi, 2023-01-02:
Hello,

thank's for the fast reply.
This initrd file in combination with the uImage file from the same commit/timespan works perfectly fine.

Thank's for the assistance!