1000001101000/Debian_on_Buffalo #168: LS-X01 Partition Help

londere, 2023-01-12:
Hello everyone, recently I get a 'damaged' LS-201 NAS, the issue was the hard drive had die, since when I power on, the device LED's start to blink. So I assume the device works, so the question is: To use Debian distro, which partition table do I need to get the device to work without use the Buffalo tool. 

In advance thanks

1000001101000, 2023-01-17:
good to hear from you!

What model is this? LS-XL, LS-210, something else?

londere, 2023-01-18:
Hello there, it's a ls210d

1000001101000, 2023-01-31:
sorry for the delays. 

The only requirements for the partition table/filesystems on that model:

- Must be GPT or MBR. (GPT secondary table needs to either be valid if using GPT and absent if using MBR).
- /boot must be on a separate partition which is relatively small ~1GB
- /boot partition must be partition #1
- /boot must be formatted as ext3
- Drive must be 16TB or less (no usually a problem)

Let me know if you have other questions.

londere, 2023-01-31:
> sorry for the delays.
> 
> The only requirements for the partition table/filesystems on that model:
> 
> * Must be GPT or MBR. (GPT secondary table needs to either be valid if using GPT and absent if using MBR).
> * /boot must be on a separate partition which is relatively small ~1GB
> * /boot partition must be partition [Network Shut Down #1](https://github.com/1000001101000/Debian_on_Buffalo/issues/1)
> * /boot must be formatted as ext3
> * Drive must be 16TB or less (no usually a problem)
> 
> Let me know if you have other questions.

[image]
Hi there, I got this over gparted section, as you can see the partition is the first and had 1024 MB named as boot, only I post this if anything is wrong, if not, I only need to put the uImage and initrd corresponding to my device, plug it to Power and Ethernet, wait until I get the IP address and start the ssh installer?

1000001101000, 2023-02-01:
That sounds correct.

londere, 2023-02-01:
> That sounds correct.

OK man I got to boot, but now I had this last problem. Hopefully the last one:

[image]

1000001101000, 2023-02-01:
When did you download the files? That typically happens when the files are too old. You might just need to download the latest installer files

londere, 2023-02-01:
It was uploaded in a page in Spanish several months ago, but I had downloaded the last update. Thanks for the support, now I had a new NAS.

closed 2023-02-01 by londere