1000001101000/Debian_on_Buffalo #169: image not recognized on TS2Pro

Jan-NiklasB, 2023-01-30:
Hey,
I stumbled over your project on the search on a way to install debian on a TS2Pro.

I followed your tutorials, so that I'm now running on stock firmware.

Now the problem:
Every time I try to boot with the switched initrd and uImage the TS shows "Lost boot image"

I tried the "Option 2" via putting it on a HDD, via ACP_Commander and TFTP.

whenever the TS restarts it shows the error and switches to TFTP.
Only when supplying the images via TFTP it boots into it, but seems to hang (no HDD actions, LINK-LED stays on even if cable is detached)

Do you have any idea how to solve this?

I'm using the bullseye img with the older initrd as written in #167 

Thanks in regard
jan-Niklas

bogatkaj, 2023-01-31:
Well,
I found my mistake on my own... I assumed that the TS will have the same IP as before, but for whatever reason it changed from 150 to 160 running the installer.

But one questing is sitll there where you could may help me...
by default the HDD is formattet with two array devices, one for boot and one for root, as well as multiple scsi partitions.

Which of them must be left as they are and which can be deleted and repartitioned?

Jan-NiklasB, 2023-01-31:
Managed to boot, but got different problem mentioned here: #170 

The right partitioning is linked in the issue mentioned above. (for everyone else experiencing trouble)

closed 2023-01-31 by Jan-NiklasB