1000001101000/Debian_on_Buffalo #170: [Bullseye-TS2Pro] Kernel doesn't support md-devices and LVM

Jan-NiklasB, 2023-01-31:
I managed to boot the current bullseye image in my TS-HTGL/R5 unsing this partitioning guide: https://github.com/DavidDah/Debian_on_Buffalo/wiki/How-to-install-Debian-on-TeraStation-PRO-II-TS-HTGL-R5

But now the installer tells me, that the current kernel doesn't supports software raind devices (mdx) and LVM...

Am I missing something or is there a bug in the current image?

1000001101000, 2023-01-31:
Good to hear from you!

mdadm should work in the installer (once it gets to the partitioner menu,  I think) but I believe I omitted the LVM related kernel modules to save space. The limitation is part of the installer image, it should work fine once you boot into debian.

The bootloader for this generation of device has some sort of a bug limiting the initrd size to 7MB, the bootloader crashes if you try to load a bigger file. The Debian netconsole installer that these images are based on is contained inside an initrd image which is subject to that 7MB limit. I've had to get creative with how I pack the image and omit certain things to keep it under the limit, even then subsequent updates of the base image sometimes make it grow and break things. 

As part of the upcoming Debian Bookworm release work I'll be re-evaluating how I construct these images to see if it's still worthwhile or if this generation device needs to be limited to the debootstrap method [https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script](url). 

If I drop support for this model in the installer I could include the LVM stuff again, but that wouldn't help for your device.

Jan-NiklasB, 2023-01-31:
Hey, thanks for the fast feedback!
I'm installing buster now, since it works fine.
The Problem with the raid is, that as far as I understand the /boot has to be on md0 (following the linked tutorial).
When i get to the partition manager in the installer the raids aren't shown, so i can't tell the installer to use md0 for /boot.

Nevertheless great work you are doing!
Maybe you or someone else will find a workaround for the limitation one day.

Keep up the great work!
Greetings
Jan-Niklas

1000001101000, 2023-01-31:
Sounds like something I'll need to look closer at. raid should work in the installer, at least well enough to look at /boot. 

I probably won't spend much time on it until I get to the Bookworm release work, but that should start fairly soon.