1000001101000/Debian_on_Buffalo #173: Which partitions to delete`

sfh1975, 2023-02-15:
Hi all
I finally decided to update my Buffalo Linkstation LS-WVL/E, due to reasons I describe below.
I already had two 4TBs WD reds and also added an old 2.5HDD via USB, running without issues.
After uploading the two recommended files, it seems to boot back into installation mode as I cannot access either root or user, but I can access installer. I cannot seem to get back to stock as well, since I dont fully understand the sequence to put it into one (the blue light flashes rapidly, function button doesnt seem to do anything). Since this drive has data, I am not sure which partitions to delete to try again the installation again. This drive has a lot of data which I dont want to lose, hence asking for your help. I am attaching a screenshot of how the first HDD looks like (I have removed the other one as well as the USB one). I have acces to an external harddrive and both linux and windows, if need be.
My questions are:
1. Can someone please guide as to which partitions to get rid of? I have no RAID, only separate drives with shares.
2. Will I be able to use more than 4TB drive in each slot? I read somewhere that the limit set by stock was 4TB in each slot and hence 8TB total. One of the main reasons I want to install debian.
3. How good or bad the performance in terms of read/ write will be, compared to stock? I only use it for storing movies and shows and use different computers in the homenetwork to access these.
4. Is it advisable to put the OS on a USB stick and run it from there? Or on a 2.5HDD plugged in the USB slot? Any pros or cons?
Thanks for responding to any or all of these Qs. :)
Faisal

sfh1975, 2023-02-16:
> Good to hear from you!
> 
> 1. For standard install partitions:
>    Looking at your screenshot you will want to tell it to format that 1GB Raid Device 0 as ext3 and set the mountpoint as /boot
>    You'll want to also format the 5GB raid device with whatever filesystem you prefer and mount as "/"
>    The others can be left as-is for now, you can deal with mounting them after the install is finished.
> 2. Drives up to 16TB should work with debian, and probably work with the stock firmware as well. I've never heard of any particular drive model that didn't work with these but there is no guarantee any particular model will work.
> 3. I don't have any data on performance for stock vs debian. I would expect for most measures stock would be faster just because it is much older software which took less memory to run than the modern version and was a bit more customized for this particular hardware. There are probably some areas where the newer versions have optimizations but the main advantage is the ability to run newer, more secure software with better compatibility with modern systems.
> 4. The boot loader for these devices won't boot directly from USB (if I remember correctly), it's also usb 2.0 which is much slower than the sata ports. You could put still put the rootfs on the USB and keep /boot on a sata drive or rely on TFTP booting etc ..... but there is little if any reason to set it up like that.

Thanks a lot and really appreciate your fast response :)
Well yesterday was a bad day as I, being the lazy ass that I am, instead of formatting the partitions as you said, deleted all partitions before the big 4TB one and then put it into the NAS itself thinking I will first restore it to stock and I will then try installing Debian again. And it did it was supposed to do :(
It formatted everything and now I have a 4TB drive with no files. Kept trying yesterday to find something to recover that partition, using a windows PC. A ocuple of free software didnt do it. Today I plan to use a linux PC and try testdisk/ photorec, lets see how it goes :(
Any recommendations for a software which might bring back the partition and almost 4TB of data back?
Thanks and much appreciate your help. Yes I know I am stupid :(
Have a nice day.
[image]

1000001101000, 2023-02-16:
testdisk is worth trying, I'm not familiar with any of the other programs out there.

sfh1975, 2023-02-19:
> testdisk is worth trying, I'm not familiar with any of the other programs out there.

Thank you for your quick reply and my late response. I am still exploring solutions, have ordered new HDDs (I needed them anyway) and will see whether I can restore patitions or have to copy files manually. Will update this thread after that, thanks again :)

sfh1975, 2023-02-24:
OK I am back :)
No good news though :(
I tried installing debian using a brand new hard disk. No luck. Couldnt log in using root or user. Only installer was working which took me back to installation. Tried marking the first partition as boot, no change. Deleted first and second partition as /boot and / but then of course it went back to installing stock OS.
No idea what to do next, really wanted to switch to debian :(
Any help before I commit to stock OS? Havent started moving files to the new HDDs. Still figuring out how to recover old HDD's XFS partition as trying to retain folder structure of files. Folks over at testdisk didnt help at all :(
Sorry for the rant!

1000001101000, 2023-02-24:
something must be incorrect about how your /boot is configured in the installer. If you try again and post a screenshot of the partitioning menu before you start the actual install I can take a look and see if I can spot the issue.

sfh1975, 2023-03-14:
Hi guys, I am back finally :)
So the buffalo is up and running. I had to download the files from Debian's server as the ones provided here are only a few KBs and obviously not the correct ones.
Anyway, now I have another issue. For the life of me, I cannot figure out how to setup samba sharing. I have done everything possible that I found online and finally reinstalled samba too, in case I had gone too far.
What I am trying to do is to share an ext4 directory containing TV shows. I want read/write access on LAN so that my other servers (Kodi, Jellyfin, Radarr, etc.,) can write to these dirs. Here is what it shows at the moment:
----------------------------------------
root@debian:~# testparm
rlimit_max: increasing rlimit_max (1024) to minimum Windows limit (16384)
Registered MSG_REQ_POOL_USAGE
Registered MSG_REQ_DMALLOC_MARK and LOG_CHANGED
Load smb config files from /etc/samba/smb.conf
rlimit_max: increasing rlimit_max (1024) to minimum Windows limit (16384)
Processing section "[TV_Shows_on_Buffalo]"
Global parameter guest account found in service section!
Loaded services file OK.
Server role: ROLE_STANDALONE

Press enter to see a dump of your service definitions

# Global parameters
[global]
        auto services = global
        log file = /var/log/samba/log.%m
        logging = file
        map to guest = Bad User
        max log size = 1000
        netbios name = BUFFALO NAS
        obey pam restrictions = Yes
        pam password change = Yes
        panic action = /usr/share/samba/panic-action %d
        passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword:* %n\n *password\supdated\ssuccessfully* .
        passwd program = /usr/bin/passwd %u
        security = USER
        server role = standalone server
        unix password sync = Yes
        usershare allow guests = Yes
        idmap config * : backend = tdb
        delete readonly = Yes

[TV_Shows_on_Buffalo]
        guest ok = Yes
        path = /media/TV Shows
        read only = No
root@debian:~# 
---------------------------------------
I have user "faisal" set up both in debian and samba. Not sure what else to do? Any ideas? :(
Thanks a lot for the help so far.
Rgds

1000001101000, 2023-03-14:
What an odd thing to claim:
[image]

sfh1975, 2023-03-14:
I am not claiming anything, I just said what I had observed :) I was right-clicking the file name and saving it. I have now realized that you have to actually click it and then download the file. Maybe I am too stupid, or maybe I am just instinctively doing what seems right to a novice like me :)
[image]

Anyway, any ideas about the samba server and file sharing? I am really struggling there :(
Thanks for your work and help as always!

sfh1975, 2023-03-17:
Thanks a lot mate. After another hiccup (the server was not booting properly after a reboot, couldnt figure out what was happening as no ssh/ webmin), unfortunately I decided to go back to the stock firmware no matter how much I wanted to use debian.
All the best for your project and thanks again.

closed 2023-03-17 by sfh1975