1000001101000/Debian_on_Buffalo #180: Is there a way to keep using the LCD and lamps with Debian?

masterlog80, 2023-05-29:
Hello,
As I am starting to swap HDD with SSD on my TS-XH2.0TL/R6, I am considering to install Debian to take advantage of properly features like `trim` or eventually `btrfs`, etc. I have already installed Debian on different models and everything was fine, but I would like keep using the display to show current information (e.g. IP, Disk usage, etc.) and the lamps for Errors, etc.
Taking a look at the OS, there is a script on */etc/cron/cron.d* calling */usr/local/bin/time_set.sh set_micon*, some other settings on */etc/melco*, but I am not really sure how LCD and lamps are managed (e.g. serial? USB?).
I am assuming Debian doesn't support those (but please correct if I am wrong), so I am wondering if someone was able to keep using those.

Regards,

1000001101000, 2023-05-29:
I created a python library for controlling the micon which includes examples:
https://github.com/1000001101000/Python_buffalo_libmicon

masterlog80, 2023-05-29:
Hello @1000001101000,
Thank you for your update and the work on this!
I don't see my device (TS-XHTL) there, but I am kinda confident it may work as well. I am not a programmer, but I am assuming it would be possibly to copy/paste the required functions from the various examples in a new file and then start in a cycle. 
I am also assuming that the LAN leds and the Power button works separately, and possibly also the FUNCTION and DISPLAY buttons.

Regards,

1000001101000, 2023-05-30:
I believe the TS-XHL is a variant of the TS-XL and should work the same

closed 2023-05-30 by masterlog80