1000001101000/Debian_on_Buffalo #182: [bookworm] Linkstation Live v2 kernel v6.1.0-9 initrd disabled, fails to find root fs, installer v5.10.136 works

digitalcircuit, 2023-06-14:
After [following the Debian guide for upgrading to `bookworm`](https://www.debian.org/releases/bookworm/armel/release-notes/ch-upgrading.en.html ), I've gotten an unbootable Linkstation Live v2.

```
[    0.000000] INITRD: 0x00800000+0x005dd000 overlaps in-use memory region
[    0.000000]  - disabling initrd

[…trimmed more messages, see full excerpt below…]

[    1.102814] List of all partitions:
[    1.106424] No filesystem could mount root, tried: 
[    1.106448] 
[    1.112894] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

<details><summary><strong>Tap/click for serial console logs: Linux kernel v6.1.0-9 initrd disabled, cannot find root filesystem</strong></summary>

```
[…]
## Booting image at 00100000 ...
   Image Name:   kernel 6.1.0-9-marvell
   Created:      2023-06-14   9:39:20 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2685139 Bytes =  2.6 MB
   Load Address: 01a00000
   Entry Point:  01a00000
   Verifying Checksum ... OK
OK
## Loading Ramdisk Image at 00800000 ...
   Image Name:   ramdisk 6.1.0-9-marvell
   Created:      2023-06-14   9:39:21 UTC
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    6146668 Bytes =  5.9 MB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK

Starting kernel ...

arg:console=ttyS0,115200 root=/dev/sda2 rw panic=5 iomem=relaxed BOOTVER=1.10
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.0-9-marvell (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 Debian 6.1.27-1 (2023-05-08)
[    0.000000] CPU: Feroceon [41069260] revision 0 (ARMv5TEJ), cr=a005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Buffalo Linkstation Pro/Live
[    0.000000] Memory policy: Data cache writeback
[    0.000000] INITRD: 0x00800000+0x005dd000 overlaps in-use memory region
[    0.000000]  - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/sda2 rw panic=5 iomem=relaxed BOOTVER=1.10
[    0.000000] Unknown kernel command line parameters "BOOTVER=1.10", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:on, heap free:off
[    0.000000] Memory: 121572K/131072K available (5134K kernel code, 846K rwdata, 1532K rodata, 320K init, 245K bss, 9500K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 24552 entries in 72 pages
[    0.000000] ftrace: allocated 72 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: orion_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 11467562657 ns
[    0.000004] sched_clock: 32 bits at 167MHz, resolution 6ns, wraps every 12884901885ns
[    0.000094] Switching to timer-based delay loop, resolution 6ns
[    0.000548] Calibrating delay loop (skipped), value calculated using timer frequency.. 333.33 BogoMIPS (lpj=666666)
[... 275 lines cut ...]
[    5.408775] ata1.00: configured for UDMA/133
[    5.414264] scsi 0:0:0:0: Direct-Access     ATA      SAMSUNG HD501LJ  0-10 PQ: 0 ANSI: 5
[    5.438700] scsi 0:0:0:0: Attached scsi generic sg0 type 0
[    5.446525] sd 0:0:0:0: [sda] 976773168 512-byte logical blocks: (500 GB/466 GiB)
[    5.459538] sd 0:0:0:0: [sda] Write Protect is off
[    5.468718] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    5.531941]  sda: sda1 sda2 sda3 < sda5 >
[    5.553031] sd 0:0:0:0: [sda] Attached SCSI disk
[    5.649183] Segment Routing with IPv6
[    5.653604] NET: Registered protocol family 17
[    5.659048] registered taskstats version 1
[    5.663248] Loading compiled-in X.509 certificates
[    5.669546] AppArmor: AppArmor sha1 policy hashing enabled
[    5.782891] ata2: SATA link down (SStatus 0 SControl 300)
[    5.807066] Freeing unused kernel memory: 340K
[    5.811576] Kernel memory protection not selected by kernel config.
[    5.817907] Run /init as init process
Loading, please wait...
Starting systemd-udevd version 252.6-1
[…]
```

</details>

For now, I can revert to Linux kernel v5.10.x to have a booting system, but this isn't sustainable long-term.  Please let me know if there's anything further that would help!

1000001101000, 2023-06-14:
Good to hear from you!

It's been taking an increasing number of tweaks behind the scenes to keep these devices booting modern kernels using their original bootloaders.

The issue that you are encountering happens because the compressed kernel unpacks itself into the beginning of memory when it starts. Modern kernels are now getting large enough that this will overwrite the initrd when loaded into it's default address. The reason this stops happening when using my kernel image is because I wrote some assembly code that I append to the front of the kernel which moves the initrd farther down in memory before starting to the kernel to prevent this.  

The easiest way to correct this might be to re-install using my bookworm installer images once I post them in the next day or two. Otherwise you could look at the install scripts in the repo and replicate the steps for enabling the "bootshim" on your system.

Another thing to watch out for with that model is a 7MB limit for the initrd (uboot crashes if you try to load a file bigger than that). I've worked around that in a couple of ways for devices using my installer process:

- Using a custom kernel which moves the drivers needed at boot time into the kernel and omitting them from the initrd
- Using XZ compression to get the initrd as small as possible

digitalcircuit, 2023-06-14:
First, **thank you for all the effort to keep these devices alive.**  The fact I've gotten around 14 years of service from this Linkstation Live v2 is great!  Both from a budget sense, and reducing electronics waste.

I appreciate your detailed explanation and solution.  In the future, I may reinstall using your upcoming bookworm installer images.

But, as I didn't want to wait nor deal with reinstalling, I've documented the steps I took in the off-hand chance it helps others who run into this issue…

1.  Boot into the installer kernel to get a working system
     * The easiest approach may be to pull the drive out of the Linkstation and [copy `uImage.buffalo.lsgl`](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Bullseye/installer_images/armel_devices/uImage.buffalo.lsgl) over top of `uImage.buffalo` in the boot partition
     * Alternatively, you can save this as `uImage.buffalo.installer`, interrupt the **second** u-Boot "Hit any key to stop autoboot" prompt (not the first one), then run `setenv kernel uImage.buffalo.installer` followed by `boot`
3.  Log in as your standard user
4.  Fetch and install the bootshim tools
```sh
cd $(mktemp -d)
# Download tools
wget "https://github.com/1000001101000/Debian_on_Buffalo/raw/master/Tools/0-install_shim"
wget "https://github.com/1000001101000/Debian_on_Buffalo/raw/master/Tools/bootshim/armel_shim"

# Copy bootshim locally
sudo cp "armel_shim" "/boot/bootshim"

# Ensure bootshim is installed with kernel updates
chmod 755 "0-install_shim"
sudo cp "0-install_shim" "/etc/initramfs/post-update.d/"

# Install bootshim into all kernels right now
sudo "/etc/initramfs/post-update.d/0-install_shim"
```
5.  Confirm that bootshim is installed
```sh
user@hostname:/tmp/tmp.a6h2T8YYQL$ sudo "/etc/initramfs/post-update.d/0-install_shim"
bootshim installed to /boot/vmlinuz
bootshim installed to /boot/vmlinuz-5.10.0-23-marvell
bootshim installed to /boot/vmlinuz-6.1.0-9-marvell
user@hostname:/tmp/tmp.a6h2T8YYQL$ 
```
6.  If you booted into the installer kernel by overwriting the file on disk (instead of temporarily modifying the u-Boot environment), reinstall the latest kernel to overwrite the installer kernel
```sh
dpkg -l 'linux-image*-marvell*'

sudo apt install --reinstall [YOUR_LATEST_KERNEL_VERSION]
```

In my case, as I am not using the custom kernel, the last two entries are…
```
ii  linux-image-6.1.0-9-marvell   6.1.27-1     armel        Linux 6.1 for Marvell Kirkwood/Orion
ii  linux-image-marvell           6.1.27-1     armel        Linux for Marvell Kirkwood/Orion (meta-package)
```
So my resulting command is…
```sh
user@hostname:/tmp/tmp.a6h2T8YYQL$ sudo apt install --reinstall linux-image-6.1.0-9-marvell
```
You should see something like this in your apt output…
```
[…]
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.1.0-9-marvell
bootshim installed to /boot/vmlinuz-6.1.0-9-marvell
Using DTB: orion5x-linkstation-lsgl.dtb
[…]
```
7.  If initrd winds up exceeding 7 MB (**untested** - my current initrd is 5.9 MB, so I am skipping this step for now), [install the custom kernel](https://github.com/1000001101000/Debian_on_Buffalo/commit/3054b78894b7dd9356d430a2fcc753c6a8f666b6#diff-bd28c581c959f5ec5031f7f0065b4cdc8785cd80ee74f3008158be11f038b403R348-R356 )…
```sh
sudo wget -O /etc/apt/custom_repo.gpg https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA/KEY.gpg
sudo apt-key add /etc/apt/custom_repo.gpg
source /etc/os-release
echo "deb https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA/ $VERSION_CODENAME main" | sudo tee "/etc/apt/sources.list.d/buffalo_kernel.list"
sudo apt-get update
sudo apt-get -y install linux-image-marvell-buffalo
```
8.  Prepare your recovery method just in case (serial console, pulling the drive out, etc), then reboot
9.  Log in as your standard user and check kernel version
```sh
uname -r
```
In my case…
```sh
user@hostname:~$ uname -r
6.1.0-9-marvell
```

1000001101000, 2023-06-15:
Nice work!

1000001101000, 2023-06-17:
The Bookworm installer files are up now.

digitalcircuit, 2023-06-17:
Thanks for the heads up!  I see that the bootshim scripts are updated, so I'll update it on my Linkstation Live. 

The one other issue I've run into so far after moving to bookworm without reinstalling is occasional brief single beeps, and this morning it started flashing error and beeping in two tones.

Though the fan seemed to be running, it felt warm.  I'm guessing I also need to set up your `micon_scripts`'s fan daemon for fan control, even though micro-evtd's config file claims it handles fan control.  Alternatively, though it didn't seem too dusty, I might just need to clean out the internals.

(I will look into reinstalling at some point to clean up the cruft from the years and ensure I've got everything as it should be.)

I'm guessing this issue can be closed, unless you'd like to keep it open as a heads up to anyone else upgrading to bookworm.

1000001101000, 2023-06-18:
if you’re using my micro-evtd binary the features from the config won’t work. looking back I should have renamed my fork of micro-evtd to something else. 

The original micro-evtd has a ton of features like fan control but is also aggressively specific to the LS-GL. 

The only feature I was interested in was its ability to send arbitrary micon messages using a small binary that could fit in a 7mb installer. So I removed basically everything else, added support for missing message types (mainly LCD stuff) and setup the build process…. I’ve found it quite amusing using it on armel,armhf,amd64 and powerpc systems.

digitalcircuit, 2023-06-18:
Ah, that's good to know.  I'm still using the original micro-evtd binary as shipped with the installer for Debian 7, possibly Debian 6, so I will **not** try to update it from this repository without also applying all the other related changes you've included.

I'll keep monitoring the beeps, and if it keeps happening I'll look into other potential causes, and simply reinstalling with your installer images.

Thank you!

digitalcircuit, 2023-07-14:
### Context for legacy `micro-evtd` trouble

**This is not an issue with your installer!**  This is a result of myself seeing how far I can get without doing a full reinstall.

Over the past month, the beeping has randomly continued, and `systemctl restart micro-evtd` works around it for an hour to a couple days.

It seems like the watchdog service part of the old (Debian 7-ish) `micro-evtd` is failing, so I've tried migrating to your simplified `micon_scripts`.

### How to upgrade to `micon_scripts`

In case this helps anyone else upgrading a legacy Debian install who doesn't want to go through the whole install process, here are the steps I took…

```sh
# Check some requirements to try to ensure a similar system
if ! command -v "micro-evtd" >/dev/null 2>&1; then
	echo "You need to have the legacy Debian 'micro-evtd' command already installed to follow this migration guide." >&2
	sleep 5
	exit 1
fi
if ! command -v "microapl" >/dev/null 2>&1; then
	echo "You need to have the legacy Debian 'microapl' command already installed to follow this migration guide." >&2
	sleep 5
	exit 1
fi

# Prepare
cd "$(mktemp -d)"

# Install Python libraries
# (These were missing from my installation - you might be missing more!)
sudo apt install python3-serial

# Download micon-scripts
mkdir "micon_scripts"
cd "micon_scripts"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/libmicon.py"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/micon_boot.service"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/micon_fan_daemon.py"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/micon_fan_daemon.service"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/micon_shutdown.py"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/micon_startup.py"
wget "https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/Tools/micon_scripts/serial_console.py"
cd ".."

# Move scripts into place
sudo cp -r "micon_scripts" "/usr/local/bin/"
sudo cp micon_scripts/*.service /etc/systemd/system/
sudo chmod 755 /usr/local/bin/micon_scripts/*.py

# Skip the shutdown service for now, keep using the legacy one
# sudo ln -s /usr/local/bin/micon_scripts/micon_shutdown.py /lib/systemd/system-shutdown/micon_shutdown.py

# Disable the legacy micro-evtd daemon
sudo systemctl disable micro-evtd.service

# Enable the new daemons
sudo systemctl enable micon_boot.service
sudo systemctl enable micon_fan_daemon.service

# Check that at least micon_boot works now
sudo systemctl start --wait micon_boot.service
sudo systemctl status micon_boot.service

# Reboot
sudo reboot
```

### Results

Services are successfully enabled and the Linkstation power LED stops blinking on boot.

I am still testing this and will edit my comment with an update on the beeping/watchdog service.

```sh
sudo systemctl status micon_boot.service micon_fan_daemon.service micro-evtd.service

# Trimmed results:
# ○ micon_boot.service - Tell Terastation it finished booting so it doesnt shutdown
#     Loaded: loaded (/etc/systemd/system/micon_boot.service; enabled; preset: enabled)
#     Active: inactive (dead) since Thu 2023-07-13 22:06:40 EDT; 3min 2s ago
#   Main PID: 316 (code=exited, status=0/SUCCESS)
#
# ● micon_fan_daemon.service - Terastation Fan Service
#     Loaded: loaded (/etc/systemd/system/micon_fan_daemon.service; enabled; preset: enabled)
#     Active: active (running) since Thu 2023-07-13 22:06:24 EDT; 3min 22s ago
#   Main PID: 317 (micon_fan_daemo)
#
# ○ micro-evtd.service - LSB: Daemon for Linkstation/Kuro micro controller
#     Loaded: loaded (/etc/init.d/micro-evtd; generated)
#     Active: inactive (dead)
```

References:
* [`Tools/debootstrap/generate_disk_image.sh`](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Tools/debootstrap/generate_disk_image.sh )
* [`Tools/late_devspec.sh`](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Tools/late_devspec.sh )
* [`Bookworm/installer_images/build/generate_images_armel.sh`](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Bookworm/installer_images/build/generate_images_armel.sh )
* [`Tools/micon_scripts`](https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Tools/micon_scripts )
* [`Tools/in_target_finish.sh`](https://github.com/1000001101000/Debian_on_Buffalo/blob/master/Tools/in_target_finish.sh )