1000001101000/Debian_on_Buffalo #184: LS500 series status?

Maximal98, 2023-06-29:
I have set up ssh access to my LS520D, only to see that it is fairly vanilla debian wheezy (while having commented out openmediavault repos, for some reason). What is currently preventing the installation of custom debian onto it?

1000001101000, 2023-06-29:
The main factor that has limited my working on it (besides time) is that the device boots from the onboard NAND which makes it more complicated to test boot modifications without the needed tools/knowledge to reflash between attempts. 

When I looked at it a few years ago I wasn't able to get a working serial console or JTAG/etc access on the board even though there were some headers on the board (Buffalo likes disabling those for some reason). There is a usb recovery feature as part of the bootloader which might help bypass some of those limitations but at the time I couldn't figure out how to actually trigger it. At the time I also wasn't sure if the SoC used by this device was sufficiently supported by the mainline linux kernel. 

Most of my time for this project since then has been focused on keeping the existing devices working with modern linux, which ended up requiring teaching myself some ARM Assembly and writing some kernel patches etc. 

Right now I'm focused on troubleshooting an issue preventing the TS3400 from running on kernel 5.18+. Once that work is done I've been thinking about taking another look at some of the newer devices without support (LS500/LS700/TS3010/TS5010) before resuming work on some of the outstanding bugs affecting a few of the older devices.

Maximal98, 2023-07-01:
Thank you for the response! I might try having a look myself, but if you get around to it i would be happy to help in testing.

Maximal98, 2023-07-01:
I've been looking into the system for some time now. I made some notes

- The "USB Recovery" image appears to be the exact same as the Firmware update image, judging by the SHA1 hash.
- The Root filesystem is a "Memory Technology Device" which is neither a block nor char device.
- MTDs are a method of interacting with raw flash memory from Linux without any memory controller or otherwise.
- I assume the firmware images are MTD images. With [a script](https://wiki.emacinc.com/wiki/Mounting_JFFS2_Images_on_a_Linux_PC#Mounting_JFFS2_Images_using_RAM) they can be mounted, however this script is made for JFFS2 Images. The firmware is not JFFS2, so we can use it as a base but it requires tweaking.
- tweaking it to use UBIFS, the alternative(?) filesystem for MTDs also fails with "cannot open "/dev/mtdblock0", error -22" (EINVAL)
- it is also not a squashfs image, which i tried because the rootfs is SquashFS'd

These are the most substantial conclusions i have drawn from my brief investigation, i hope they can be of any help.

1000001101000, 2023-07-01:
I do have some notes on the wiki about the device and running custom code:
https://buffalonas.miraheze.org/wiki/Linkstation_LS520D

I did include support for making a debian image you can run within the stock firmware on the device, I think the last version I actually tested that with was Stretch:
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script

Maximal98, 2023-07-01:
oh, i oversaw your existing notes, sorry about that. However i was already aware of the chroot method, but just curious about the installation of a custom OS

Maximal98, 2023-07-02:
The firmware "image" is just a renamed tar file, probably to throw people off, and also probably worth a note on the wiki