1000001101000/Debian_on_Buffalo #185: Unable to boot LS220D installer images

puzzled55, 2023-07-07:
Great work on this project.
I was having trouble with the installer images for LS220D (using single drive1), both for Bullseye and Bookworm.
I checked the method works for the standard firmware files eg v168 and that was fine i.e. https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP
But using that method for the Debian installer images doesn't work.
I notice the initrd.buffalo for the installer images are noticeably larger than the ones in the standard fw - could that contribute to the issue?
But the good news is that this method works
https://github.com/1000001101000/Debian_on_Buffalo/wiki/Alternate-install-method-via-debootstrap-script

1000001101000, 2023-07-07:
For this model the size of the files isn't important like it is for the older models. 

When you say they didn't work, how did you determine that? What did the LEDs on the front of the device do when you tried to boot them?

Did the device have access to the internet while it was trying to boot the installer files?

puzzled55, 2023-07-07:
When it didn't boot - the LEDS were
top/power - flashing white 
lower/function - off
The power led was flashing for the whole time I left the unit on for up to an hour before I gave up. I took the drive out and checked it afterwards - the two xxx.buffalo files were still there and nothing changed. (I later placed FW163 xxx.buffalo files in there and unit booted just fine - into EM so that I can use LSUpdater etc)
Although the unit was connected to a router - it never connected to the internet because it never booted to request DHCP etc (no lease on the router).
By the way - I know how to use the ssh@installer method as I have tried that before with another device - an LS-WXL (LEDs are a little different on that model).

1000001101000, 2023-07-07:
could you post the size and md5sum of the files you are trying to boot with?

These are the ones I just got from the repo and successfully booted my ls220d with:
```
$ du -h uImage.buffalo.ls220d initrd.buffalo
5.2M	uImage.buffalo.ls220d
17M	initrd.buffalo
$ md5sum uImage.buffalo.ls220d initrd.buffalo
db6b0f259e14eb796f8e7846f3d3f7d7  uImage.buffalo.ls220d
b50e0748ae35ae6e9e693c5b383203ab  initrd.buffalo
```

1000001101000, 2023-07-07:
I think I see the issue now, if you download the new installer files that were generated last night it should work.

puzzled55, 2023-07-07:
Thanks for checking and finding the issue.
These were files I had downloaded previously
afe288b3613d7cc6cbb4e1ff3492556d  lswxl-bookworm/initrd.buffalo
b907a53b5b316a3956299522eb415708  lswxl-bookworm/uImage.buffalo.lswxl
74e591f668b0c92e89c717f1d9480c1e  ls220-bookworm/initrd.buffalo
e47902ff10b3579b8b79d9a57d611f32  ls220-bookworm/uImage.buffalo.ls220d

484f05223523ebc61f6c5f8591d3d7a0  lswxl-bullseye/initrd.buffalo
22a5fde6cd98a32d7091b2a90f90b0ee  lswxl-bullseye/uImage.buffalo.lswxl
617e469efa9c4b9deb2b7485a02962d8  ls220-bullseye/initrd.buffalo
11d0844677a63a3b741863b1c66d191a  ls220-bullseye/uImage.buffalo.ls220d

d86e8dedb9daf86a053f6d171914888a  lswxl-buster/initrd.buffalo
6ec313c77392c9a93c6918438ef5142a  lswxl-buster/uImage.buffalo.lswxl

These are the ones from the git clone  this morning
bookworm
armel
c33133f165c9e0097ebb271ea1855d36  initrd.buffalo
b907a53b5b316a3956299522eb415708  uImage.buffalo.lswxl
armhf
b50e0748ae35ae6e9e693c5b383203ab  initrd.buffalo
db6b0f259e14eb796f8e7846f3d3f7d7  uImage.buffalo.ls220d

bullseye
armel
7376717443126a6f0ad742ba40a6a657  initrd.buffalo
1005bf712f3e6c7ef23d2655b4f166b4  uImage.buffalo.lswxl
armhf
c605fbd57260867dd8d3bd515202b0af  initrd.buffalo
52f25e08b3c029eb1399c76242e4a346  uImage.buffalo.ls220d

closed 2023-07-08 by puzzled55