1000001101000/Debian_on_Buffalo #186: Some findings/details on the wiki

smsimeonov, 2023-07-14:
Thanks for the project, first of all. 
I'm in the process of reviving 4 old Buffalo NAS I have and handing those off to my relatives. These are two LS-QL, one TS-HTGL v2 and a HS-DHTGL.

Initially I tried just pushing a merge request, but its been a while I've collaborated on a project lol (I have my own private GitLab)

In relation to transferring uImage and initrd, I wanted to add the following after the xfer acp_commander lines:

NOTE: If the above commands complete too fast, make sure the transfer actually happened. If you do not have permission to write to /boot, files "initrd.buffalo" and "uImage.buffalo" already exist or you simply don't have a /boot - ACP Commander will soft fail without any message. You can enter a "shell" of sorts on stock firmware by using the -s flag. That way you can check whether the transfer happened reliably.

Aside from this, I found that all the devices I have (all armel), can be forced to boot to EM via just removing all drives and starting without them. The only difference being (from EM boot via acp_commander or button combination) that the hard drives take longer to spin up initially (around 15 seconds). My LS-QL had no other options for EM boot aside from that and a serial connection. Considering I have to resolder some things on the mainboard for the latter to work, I skipped it of course.
I might try to add this to the Miraheze wiki too.

It probably is worth it to mention in the wiki another good source for info, which is the old nas-central.org site, which could still be accessed via archive.org (https://web.archive.org/web/20190302093121/http://buffalo.nas-central.org/wiki/Main_Page). That obviously doesn't appear in searches anymore, so it would be good to note imo.

Last I have some notes on trying to optimize file transfer speeds.
Since all 4 devices I have are quad drive models, I set up softwareRAID (mdadm) as RAID5 on all. This has a huge impact on performance, but I mostly need the capacity as long as the read rate is above 7-8 Megabytes per second.
This works with all types of transfer methods (Samba, NFS, FTP, FUSE). They all get very similar read speeds (around 14-20 Mgbps).
The issue comes with write speeds. All new homes of these NAS have had their media archive on a PC, so a part of the initial setup is transferring them over to the NAS. Since we're talking about TB worth of data, the faster it is - the better. 
Depending on the device, the write speeds were between 4 and 7 Mgbps, but this is through SMB. NFS and FUSE returned the same speeds, while FTP writes were better with speeds generally between 11 and 14Mgbps writing.
The biggest bottleneck appears to be the CPU. In the case of SMB/NFS/FUSE, around 50% of the CPU was for their respective process, while the other 50 was for mdadm. With FTP, the proportion was around 85% to 15%.
I have tested the arrays themselves with fio in all scenarios applicable and the RAID itself is capable of at least 4 times the bandwidth it is given by the file transfers.

As one issue, possibly with the kernel, is that on the LS-QL's, the USB ports do not seem to be working reliably. They would error out with connerr 22 (via journalctl) randomly, even when not in use.

I will add any other findings as I go through further with the process.

1000001101000, 2023-07-18:
howdy:

I have a link to the internet archive of nas-central on the main page of the wiki on miraheze.

I have made sure the XOR acceleration features of the SoCs are enabled wherever possible…. but to be honest I’ve never looked into how to actually use them for raid5. There’s probably a number of optimizations possible for some of these devices.

For the LS-QL I recall I had to patch the kernel to allow enabling one of the usb ports, I seem to remember one or both were also unreliable. I also think I found it to have poor performance even for that generation. There is probably some kernel work that still needs to be done for that model.

smsimeonov, 2023-07-18:
Thanks for your input!

Since I am not familiar with XOR acceleration, I will look up some details on it to see if I can get better write rates at least.

And for the LS-QL, I ended up setting up a SFTP file backup to one of my public facing servers overnight to have a [centralized] backup. I figured out it is a kernel issue, or maybe the microcontroller itself. I noticed there is a fluctuation on the voltage output which I haven't see on other systems I have. That could or could not be the reason. I tested since I have plenty of SBC's and often when I get that error (connerr 22) it is because there is not enough power to the USB device (Connecting something with a high power draw to a USB 2.0 port, Power supply to the SBC is insufficient or in one case - needing a kernel patch)

1000001101000, 2023-07-18:
For clarity: What OS are you running on your LS-QL? If you post your DMESG output I can take a quick look.

I would expect it to load at boot since it's enabled by default for all orion5x devices:
```xor: dma-controller@60900 {
				compatible = "marvell,orion-xor";
				reg = <0x60900 0x100
				       0x60b00 0x100>;
				status = "okay";

				xor00 {
				      interrupts = <30>;
				      dmacap,memcpy;
				      dmacap,xor;
				};
				xor01 {
				      interrupts = <31>;
				      dmacap,memcpy;
				      dmacap,xor;
				      dmacap,memset;
				};
			};
```
from: https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/marvell/orion5x.dtsi

I don't have an orion5x device online at the moment but I would expect something similar to what I see on the TS-XEL:
```
[    1.238644] kirkwood-pinctrl f1010000.pin-controller: registered pinctrl driver
[    1.241439] mv_xor f1060800.xor: Marvell shared XOR driver
[    1.277554] mv_xor f1060800.xor: Marvell XOR (Registers Mode): ( xor cpy intr )
[    1.281653] mv_xor f1060900.xor: Marvell shared XOR driver
[    1.317595] mv_xor f1060900.xor: Marvell XOR (Registers Mode): ( xor cpy intr )
```

smsimeonov, 2023-07-18:
I am running your Debian build for Bullseye, since Bookworm install didn't work for me.

`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.179 (root@0f8a322422df) (arm-linux-gnueabi-gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #17 Wed Jun 14 17:41:29 UTC 2023
[    0.000000] CPU: Feroceon [41069260] revision 0 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: Buffalo Terastation Pro II/Live
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Clearing invalid memory bank 0KB@0xffffffff
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x00000000
[    0.000000] Ignoring unrecognised tag 0x41000403
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 bootshim=1
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] Memory: 112856K/131072K available (7569K kernel code, 1115K rwdata, 2144K rodata, 380K init, 233K bss, 18216K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 32833 entries in 97 pages
[    0.000000] ftrace: allocated 97 pages with 3 groups
[    0.000000] NR_IRQS: 16, nr_irqs: 65, preallocated irqs: 65
[    0.000000] Switching to timer-based delay loop, resolution 6ns
[    0.000016] sched_clock: 32 bits at 166MHz, resolution 6ns, wraps every 12884901885ns
[   19.457526] clocksource: orion_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 11467562657 ns
[   19.457761] Calibrating delay loop (skipped), value calculated using timer frequency.. 333.33 BogoMIPS (lpj=666666)
[   19.457807] pid_max: default: 32768 minimum: 301
[   19.458781] LSM: Security Framework initializing
[   19.459176] Yama: disabled by default; enable with sysctl kernel.yama.*
[   19.459605] AppArmor: AppArmor initialized
[   19.459874] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[   19.459915] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[   19.462726] CPU: Testing write buffer coherency: ok
[   19.465344] Setting up static identity map for 0x8200 - 0x823c
[   19.467321] devtmpfs: initialized
[   19.469822] VFP support v0.3: implementor 41 architecture 1 part 10 variant 9 rev 0
[   19.470114] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[   19.470165] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[   19.470580] pinctrl core: initialized pinctrl subsystem
[   19.474458] NET: Registered protocol family 16
[   19.475584] DMA: preallocated 256 KiB pool for atomic coherent allocations
[   19.480749] audit: initializing netlink subsys (disabled)
[   19.482686] thermal_sys: Registered thermal governor 'fair_share'
[   19.482703] thermal_sys: Registered thermal governor 'bang_bang'
[   19.482732] thermal_sys: Registered thermal governor 'step_wise'
[   19.482756] thermal_sys: Registered thermal governor 'user_space'
[   19.482825] cpuidle: using governor ladder
[   19.482861] cpuidle: using governor menu
[   19.483006] Orion ID: MV88F5281-D0. TCLK=166666667.
[   19.483092] Orion: Applying 5281 D0 WFI workaround.
[   19.483226] initial MPP regs:
[   19.483240]  44440000
[   19.483254]  00000000
[   19.483273]  00000000
[... 322 lines cut ...]
[   56.705613] XFS (md3): Ending recovery (logdev: internal)
[   56.764246] EXT4-fs (md0): mounting ext3 file system using the ext4 subsystem
[   56.874673] EXT4-fs (md0): mounted filesystem with ordered data mode. Opts: (null)
[   56.898752] xfs filesystem being mounted at /mnt/array1 supports timestamps until 2038 (0x7fffffff)
[   56.911599] ext3 filesystem being mounted at /boot supports timestamps until 2038 (0x7fffffff)
[   58.109696] audit: type=1400 audit(1689280596.684:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe" pid=387 comm="apparmor_parser"
[   58.157072] audit: type=1400 audit(1689280596.684:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="nvidia_modprobe//kmod" pid=387 comm="apparmor_parser"
[   58.217091] audit: type=1400 audit(1689280596.792:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/sbin/haveged" pid=388 comm="apparmor_parser"
[... cut ...]
[   76.249892] audit: type=1326 audit(1689280614.824:16): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=/usr/sbin/haveged (enforce) pid=487 comm="haveged" exe="/usr/sbin/haveged" sig=31 arch=40000028 syscall=122 compat=0 ip=0xb6f9b72c code=0x80000000
[   77.591079] audit: type=1326 audit(1689280616.164:17): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=/usr/sbin/haveged (enforce) pid=491 comm="haveged" exe="/usr/sbin/haveged" sig=31 arch=40000028 syscall=122 compat=0 ip=0xb6f5872c code=0x80000000`

smsimeonov, 2023-07-18:
> Are you booting a disk you originally set up on your TS-HTGL or something like that?

Sorry, I forgot I had the TS-HTGL plugged right now.
I am absolutely sure it was the same case with the LS-QL, but I can check later.

Regardless, shouldn't this appear on the TS-HGTL being that its an Orion platform too?

1000001101000, 2023-07-18:
The TS-HTGL or ts2pro is a different Orion5x SoC and is also defined a bit differently in a way that pre-dates device-tree.

I remember when I added support for the XOR and/or CESA engines for the TS-XL I noticed that kernel code for the ts2pro lacked some entries. At the time I looked at the SoC docs and determined that SoC didn’t support it. I think I even discussed it with whoever had an issue open at the time.

Looking at the kernel code I don’t see entries for either:
https://github.com/torvalds/linux/blob/master/arch/arm/mach-orion5x/terastation_pro2-setup.c

smsimeonov, 2023-07-18:
So I checked the two LS-QL dmesg logs and both of them have mv_xor. Very strange as I am getting even worse performance on them than compared to the TS-HTGL. I guess I will look into it further. 

Attaching the dmesg output. It even has the USB errors I mentioned.

[LS-QL_DMESG_updated.txt](https://github.com/1000001101000/Debian_on_Buffalo/files/12088274/LS-QL_DMESG_updated.txt)

1000001101000, 2023-07-18:
I'm vaguely aware that the SoC of the Terastation was one of the high-performance versions compared to the more budget one in the LS-QL but don't really know any details about all that. The 4x SATA ports via a PCIe sata controller on the Terastation vs the port multiplier connecting the for bays to a single SoC SATA port on the LS-QL will have impact on throughput etc but again, I'm not really versed on the specifics. 

Still, best I can tell the LS-QL is slower than the other Orion5x Linkstations, possibly beyond what would be expected by the hardware alone,  and I don't have a good theory as to why.

smsimeonov, 2023-07-18:
You've been very helpful, thanks!
One last thing if you can:
I cloned the repo. If I were to build my own kernel, I would just substitute the config with one of the mainline configs as how the note in your folder structure mentioned, correct?

1000001101000, 2023-07-18:
you could probably adjust things to match what makes more sense to you.

To build a kernel with my scripts:

- use update_kernel_source.sh to download the kernel source and apply my patches to it. (or do the equivalent manually)
- load the appropriate base kernel config in the proper spot
- run build_kernel_armel.sh to compile kernel packages.
- if your changes affect dts/dtsi files and the dts for your device isn't already in the mainline kernel source, load your dts files into dts/<kernel_version>. then run build_dtb.sh to generate new dtbs in dtb/<kernel version>/