1000001101000/Debian_on_Buffalo #189: Hello, can you upload the installer_images for debian 10 again.

cash1674, 2023-07-22:
First of all, thank you for your hard work. I was going to install it on TS1400D today, only to find that the installer_images of debian 10 is no longer there. All my current devices have been running stably on debian 10 for 3 years, and I am very satisfied with the current situation. Debian11 is not friendly to hardware requirements. Can you resume uploading debian10 again, this time I will download the backup and save the image for later useI'm happy with the current state, very stable, debian 11 is not very friendly and difficult to adapt to。

cash1674, 2023-07-22:
Thanks again, waiting for your good news

1000001101000, 2023-07-24:
Talked about options over on discord

closed 2023-07-24 by 1000001101000