1000001101000/Debian_on_Buffalo #190: Buffalo LS210 bookworm not working

SigmaticThar, 2023-07-28:
Hi
I grabbed the files from here https://github.com/1000001101000/Debian_on_Buffalo/tree/master/Bookworm/installer_images/armhf_devices

initrd.buffalo
uImage.buffalo.ls210d  >>>renamed>>> uImage.buffalo 

I reverted back to my Bullseye files and they work. If these files do indeed work is it possible to get a sha hash of them so I can check my files?

1000001101000, 2023-07-28:
when did you download them?

SigmaticThar, 2023-07-28:
Thanks for getting back to me. I notice they were updated three days ago and I download before that so it's probably the issue. Thanks for the hashes I'll download the new files check the hash and give it a go. I'll report back either way but it may be Sunday before I try this. :)

SigmaticThar, 2023-07-29:
@1000001101000 
I decided to do this tonight and the install was successful.
Thanks for the help!

closed 2023-07-29 by SigmaticThar