1000001101000/Debian_on_Buffalo #192: Question: Overclock CPU

Kulturnilpferd, 2023-08-26:
Hey there,
I lover really your implementation but I have also seen that the ls220d i have has the same CPU like the LS441D. But it is clocked as 800 MHz instead of 1,2GHz from the LS441D. Is there some chance to crank up that CPU by editing the boot image or firmware? This would be awesome =)

1000001101000, 2023-08-27:
I briefly looked at the documentation for that SoC, it didn't look like there was anything as straightforward as messing with the FSB/multiplier like with a PC. There is a core clock and some PLLs that can be set, I imagine some combination could be used to change the CPU speed.

The relevant documentation can be found here:
http://web.archive.org/web/20150508181644/http://www.marvell.com/embedded-processors/armada-300/assets/ARMADA370-FunctionalSpec-datasheet.pdf

It's normally not the kind of hack I'd spend any time on. Getting Bookworm to work for the TS3400 ended up requiring messing with some PCIe registers in a way I normally would stay away from, so I do have some experience with similar tweaks now. 

I'm somewhat tempted to try underclocking an LS400 to match the LS200 by manipulating the relevant registers to see if that's even feasible.

Kulturnilpferd, 2023-08-27:
This would be awesome, have also tried to just change some variables in /sys/ and play around with the kernel but have sadly nothing that really works but I'm still experimenting...

Kulturnilpferd, 2023-08-31:
Hey there, I have found this maybe it will help to get it right:
https://linux-arm-kernel.infradead.narkive.com/dgFnAfVq/patch-01-10-arm64-marvell-enable-armada-3700-clock-drivers

Kulturnilpferd, 2023-09-03:
Little update:
I have now gathered the right document to implement it to the .dts file:
https://www.kernel.org/doc/Documentation/devicetree/bindings/clock/mvebu-core-clock.txt

Now I must found out how to implement it right and compile the dts...

1000001101000, 2023-09-03:
You've correctly identified that the clocks you are interested are detailed on that page and that they are mapped to certain pin inputs which are presumably hardwired on the board. I would think this means that there would be no way to change those clocks via device-tree or other software means. Page 80 goes into a little more detail about how changing those values take effect when the CPU is reset.

Before you start moving pullup/pulldown resistors around you should double check the specs on the RAM chips since it looks like those clocks get set together.

Kulturnilpferd, 2023-09-04:
Not really sure about doing this because I have no spare hardware if something get messed up and with no circuit diagram it is more like a try and error. I could also desolder the CPU to get the right pins and where they go to the right resistors but this is too much effort to get +400Mhz more speed so if this is not working with software I would sadly close this little project.
I'm just curious how that +400MHz would have affected the file transfer rate...

1000001101000, 2023-09-04:
I have a few messed up devices in my parts bin that can be fixed via that kind of soldering but currently I have neither the tools or the skills for it.

I’d be somewhat interested to see something like that since folks are hitting CPU/Memory bandwidth limits rather than IO limits for these devices.

I suspect there are better avenues for performance gains via some software optimizations first.

Kulturnilpferd, 2023-09-09:
Hm, I could really think that the software is not the problem because if you execute
`dd if=/dev/zero of=/dev/null`
the write speed is the same as copying things to the nas per samba or anything else. Really don't know if some kernel optimizations would get a better read/write performance with that low hardware specs...

1000001101000, 2023-09-09:
Add “bs=4k” to your dd command and try again

Kulturnilpferd, 2023-09-09:
Wow, ok 82MB vs 404MB.

1000001101000, 2023-09-10:
Yup, might be faster with an even higher value. Depends on the specifics if the hardware you’re reading, anything less than a memory page and/or memory cache line will be greatly inefficient.