1000001101000/Debian_on_Buffalo #194: bullseye-backports 6.1 kernel bootloops LS-WVL

alvinhochun, 2023-08-30:
Just want to write this down somewhere public. OpenMediaVault enables kernel backports by default, which is something I did not know before. One day `apt upgrade` installed `linux-image-6.1.0-0.deb11.9-marvell` and I did not pay attention, so in the end I had to get myself a SATA dock to fix it.

I have no idea what causes that kernel to bootloop.

1000001101000, 2023-09-12:
Hard to say or sure without serial console output. My guess would be it's less about the kernel itself and more something it changed that effected the initrd or fstab/etc.

alvinhochun, 2024-12-08:
Update: I decided to try upgrading to Bookworm. While doing so I downloaded the new DTB from this repo. I wondered if the old DTB I downloaded for Buster caused the issue with bullseye-backports, but it seems to be identical to the one for Bookworm...

Anyway, I now appear to be running smoothly on Bookworm with OMV 7. So, it works on the Bookworm 6.1 kernel, but somehow not on Bullesye backports. I guess I will have to pay attention when the Bookworm backports kernel arrives in the future.

By the way, is there any reason to continue using the DTB from this repo as opposed to the one that comes with Debian? I think the biggest issue was the fan control being inverted. Do you know, has it been fixed upstream or is it still a problem?