1000001101000/Debian_on_Buffalo #195: Failed to create a file system

MelteaMoon, 2023-09-03:
Hello,

my current NAS is the Buffalo Linkstation LS-WVL. I can not install debian correctly on it, unfortunately. I already tried about 6 times, once with an "oldstable" version. It fails every time with the creation of the ext4 file system on the 4TB partition (I followed the every step of https://www.youtube.com/watch?v=dfMHlLmsSF8 this video).
The debian installation, however, is successful. But, upon executing "apt update", it fails due to a deprecated key for this repository, I guess?
(I am a bloody beginner, therefore sorry for the unclear desriptions. I will add the fail-message from "apt update" later since I already restarted the installation and have to wait for it to finish)
I have two 4TB drives and would like to set them up as RAID 0 both of my drives are Ironwolf Pro 4TB. (I also failed to somehow set them both up so that the NAS uses both, so I would appreaciate help there too.)

(IF POSSIBLE: I would like to use the USB port and install the boot and root partition on a USB drive, so that the nas does not need to boot from the HDD drives themselfs, but can use the whole drives solely for RAID 0 storage.)

I would appreaciate every help, since the wiki and the video only cover the basics and as a beginner, I am pretty lost, thank you.

(P.S. I have been wondering. Is it unhealthy  for the drives to be formated and reinstalled again so many times? I already tried about 6 times. So I am concerened, especially since I just bought those 2 drives a couple of days ago.)

Drives: 2 x Ironwolf Pro 4TB 
NAS: Buffalo Linkstation LS-WVL (LS-WV4.0TL/R1-EU)
Stable Debian (12): Bookworm
Oldstable Debian (11): Bullseye
Method used: Option 1 - Load onto Device Running the Stock Firmware

MelteaMoon, 2023-09-03:
Okey, the following:
On "oldstable" I can not reacreate the problem with the deprecated key. It seems to be an issue starting with Bookworm.
The problems with the ext4 formating is still there.

I will try to set everything up the best I can, until I receive some help.

1000001101000, 2023-09-03:
the deprecation warning about the repository key isn't an error message, just a warning. 

What exactly are you trying to accomplish? (what do you want to happen and what commands are you entering to accomplish it)

1000001101000, 2023-09-12:
@MelteaMoon Was there anything else you wanted me to look at for this issue?

closed 2023-09-21 by 1000001101000

MelteaMoon, 2023-09-28:
I am sorry,
I forgot to check my messages, I have been rather busy over the last month.
Still, thank you for your fast reply.