1000001101000/Debian_on_Buffalo #196: Latest Installer Images Not Working for TS2Pro on Bookworm and possibly Bullseye

williamcfarmwise, 2023-09-06:
Appears that checkin 22796ef1 has broken the installer image for TS2Pro (don't know about others, as I only have a TS-Live to test with, which is the same as a TS2Pro hardware-wise). I get System Error E04 Can't Load Krnl error when attempting to use the image to install. I reverted to checkin d91c81d to get the bookworm installer image to work. Latest Bullseye installer image also appears to have the same symptom, but I did not check which previous version worked, but I suspect the same.

1000001101000, 2023-09-06:
Looks like a silly mistake on my part. 

I've:
- recreated the issue on my ts2pro.
- added some logic to the installer startup to fix it and prevent me making that same mistake in future
- retested on my ts2pro.

There's not much reason to repeat your install if you ended up with a working system but you should copy the latest `micro-evtd-armel` from the repo into `/usr/local/bin/micro-evtd` on your device. Ironically, the update that broke these installer images was mainly targeted at making the boot process more reliable in ways that affect the ts2pro.

williamcfarmwise, 2023-09-07:
Thanks for the update, the new initrd.buffalo has fixed the krnl error. I can now boot into the installer. Upon further testing I found that during the partitioning portion of the installer, get the following error: "Software RAID not available, The current kernel doesn't seem to support software RAID (MD) devices. This should be solved by loading the necessary modules." I didn't see that error in the previous working install, so assuming that this is another issue in the latest image. Please let me know if you can reproduce. Thanks again.

1000001101000, 2023-09-12:
I just gave it a try on mine and was able to create a RAID1 device without issue. 

Can you provide more detail of the steps you are taking and the raid level/etc you are using? Also please post a screenshot of the error.