1000001101000/Debian_on_Buffalo #199: Resize Boot/Root Partition after raid configuration

MelteaMoon, 2023-09-28:
Hello,

I really new to this and have a problem I am afraid to work on.
During the setup on my LS-WVL I use the default partitioning, now it seems that the partition is full, even though I only installed things like samba, webmin, wsdd and a firewall.

Means I can not update nor install new software like a NFS-Server.
Is there a way for me to resize the root/boot partition from 5GB to at least 10 GB, without losing my data?

I am running two IronWolf Pro 4TB in Raid0. And I don't have enough offline storage to save all the data. (The most important ones are already backed up)

And It would be a real pain to download all those files again.
(Would be a total of almost 3 TB, and I am downloading those files to my PC and then copying them over my network to my nas, since my linkstation has a really slow download speed over the internet, for whatever reason.
Works faster this way, however, no automation possible. Therefore a huge pain to do again.)

Thank you in advance!

1000001101000, 2023-09-28:
Could you post the output of:
df -h
lsblk

MelteaMoon, 2023-09-28:
df -h:

Filesystem      Size  Used Avail Use% Mounted on
udev            119M     0  119M   0% /dev
tmpfs            26M  7.6M   19M  30% /run
/dev/md1        4.7G  4.6G     0 100% /
tmpfs           122M     0  122M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
/dev/md0        952M   31M  873M   4% /boot
/dev/md127      7.3T  2.4T  4.5T  35% /mnt/raid0
tmpfs            25M     0   25M   0% /run/user/1000

lsblk:

NAME      MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda         8:0    0   3.6T  0 disk
├─sda1      8:1    0   977M  0 part
│ └─md0     9:0    0 976.9M  0 raid1 /boot
├─sda2      8:2    0   4.8G  0 part
│ └─md1     9:1    0   4.8G  0 raid1 /
├─sda3      8:3    0     1M  0 part
├─sda4      8:4    0     1M  0 part
├─sda5      8:5    0   977M  0 part
│ └─md10    9:10   0   977M  0 raid1 [SWAP]
└─sda6      8:6    0   3.6T  0 part
  └─md127   9:127  0   7.3T  0 raid0 /mnt/raid0
sdb         8:16   0   3.6T  0 disk
├─sdb1      8:17   0   977M  0 part
│ └─md0     9:0    0 976.9M  0 raid1 /boot
├─sdb2      8:18   0   4.8G  0 part
│ └─md1     9:1    0   4.8G  0 raid1 /
├─sdb3      8:19   0     1M  0 part
├─sdb4      8:20   0     1M  0 part
├─sdb5      8:21   0   977M  0 part
│ └─md10    9:10   0   977M  0 raid1 [SWAP]
└─sdb6      8:22   0   3.6T  0 part
  └─md127   9:127  0   7.3T  0 raid0 /mnt/raid0
mtdblock0  31:0    0   384K  1 disk
mtdblock1  31:1    0    64K  1 disk
mtdblock2  31:2    0    64K  0 disk

1000001101000, 2023-09-28:
RAID0 doesn’t leave you any good options for messing with your main data array to make space for a bigger rootfs. You can steal some from /boot though that may not be worth the effort.

You should start by cleaning up some things to free some space, from there you can look around at what is taking up all the space and look at moving some things to the data array etc.

I’d start with:
apt-get clean
apt-get autoclean
apt-get autoremove

That should get some space cleared up, then I usually poke around with:
du -shBM | sort -h

MelteaMoon, 2023-09-28:
Thank you for the fast reply.
I tried the commands above, and I already used autoremove before, with no luck.
Nothing has been deleted / cleared.
Would you recommend reducing / using the swap partition?
(I am thinking of uninstalling webmin tbh. I am not really using it. And if I need a terminal I use putty or an ssh-app. 
Although, the graphical overview is handy. I would love to find an alternative for that.)

1000001101000, 2023-09-29:
I would focus on using du/etc to find where all the space is being used and address whatever the cause is…. Clean up the logs/downloads/etc.

It would be a fair amount if work to rearrange things to take space from /boot and swap just add another gig to the rootfs. Odds are it would end up using that space too until you address the underlying cause.

MelteaMoon, 2023-10-02:
I found the issue I believe.

I continued to search a bit around in my filesystem as you suggested with du -hxd.

Turns out that in /var/log/minidlna there where log files which had sizes up to 1 GB.
I think during my minidlna setup I enabled the logs, thinking they would only eat up maybe 150 MB, believing really old logs would be deleted automatically.
Well, oopss... ^^"
I uninstalled minidlna and deleted the log files. I now have 3.0 GB free again for my use, I will set up minidlna on a separate device. That should solve the issue and reduce the burden on my old nas.

Thank you for your help and tips!

closed 2024-02-01 by 1000001101000