1000001101000/Debian_on_Buffalo #201: repo error

jhmc93, 2023-11-10:
I have debian bookworm on my terrastation and every time i run apt update it comes up with the following:
"W: https://raw.githubusercontent.com/1000001101000/Debian_on_Buffalo/master/PPA/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details."
any ideas on how to fix?

1000001101000, 2023-11-10:
It's a warning which I haven't gotten around to looking into. Here's the documentation referenced by the warning:
https://manpages.debian.org/testing/apt/apt-key.8.en.html#DESCRIPTION

jhmc93, 2023-11-10:
Is there a way to disable the warning

1000001101000, 2023-11-11:
I’m not planning to work on it until closer to the Trixie release.  The page I linked did mention what the new process is, that might make it go away.