1000001101000/Debian_on_Buffalo #202: Driver installation issue

MelteaMoon, 2023-11-23:
Hello again,

I am trying to install a WIFI driver from source. Problem is, some directory / file is missing and every attempt to fix it failed and every reseach led to no result.

Below the output after the failed "sudo make install / sudo make" command: (can not remember which of these two it was.)

DKMS make.log for rtl88x2bu-5.8.7.1 for kernel 5.10.197 (armv5tel) Mon 20 Nov 2023 04:08:34 AM CET make ARCH=armv5tel CROSS_COMPILE= -C /lib/modules/5.10.197/build M=/var/lib/dkms/rtl88x2bu/5.8.7.1/build modules make[1]: Entering directory '/usr/src/linux-headers-5.10.197' Makefile:714: arch/armv5tel/Makefile: No such file or directory make[1]: *** No rule to make target 'arch/armv5tel/Makefile'. Stop. make[1]: Leaving directory '/usr/src/linux-headers-5.10.197' make: *** [Makefile:2392: modules] Error 2

I hope you could help me with this. I wanted to see if my wifi stick would be faster than the internal LAN chip.

Thanks in advance!

1000001101000, 2023-11-23:
changing to `ARCH=arm` should fix the error you are seeing.

I can not think of a scenario where a wifi dongle over USB 2.0 (480Mb/s max) would be faster than the onboard 1Gb/s NIC. Still, there are plenty of reasons a wifi connection could be useful.

MelteaMoon, 2023-11-24:
Well, first of all I did not know it was a 1 Gb/s NIC. However, I definitely do not read / write off / to my NAS with 1 Gb/s.
I can understand if the writes are low, makes sense. But the reads are almost the same as the write speed. 
My writes are at best a constant 28 - 30 MBytes/s, and the reads were once or twice almost a 100 but most of the time it is at best 35 MBytes/s. (And I am transferring mostly large Movie files or larger compressed files. Most of the time above 20 GB) 
And I do not know why, maybe I set the wrong blocksize / settings? 
It is just really weird.
But thank you for the help.
I will tell you if it worked or not.

MelteaMoon, 2023-11-24:
It did not fix it. Or at least there is the same error cause by something different.

make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.197/build M=/home/user/RTL88x2BU-Linux-Driver  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.197'
  CC [M]  /home/user/RTL88x2BU-Linux-Driver/core/rtw_cmd.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:291: /home/user/RTL88x2BU-Linux-Driver/core/rtw_cmd.o] Error 126
make[2]: *** Deleting file '/home/user/RTL88x2BU-Linux-Driver/core/rtw_cmd.o'
make[1]: *** [Makefile:1856: /home/user/RTL88x2BU-Linux-Driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.197'
make: *** [Makefile:2470: modules] Error 2

I really do not get it.

1000001101000, 2023-11-24:
You'll either want to compile it on the device itself or use a cross-compiler, that error indicates it's trying to build an arm binary with a non-arm toolchain... I think.

The speed of the NIC is one of several factors that dictates how fast a file transfer will go. You can us ethtool to see what speed the NIC is running at and something like iperf to measure raw network performance to/from another system.

MelteaMoon, 2023-11-25:
Thank you, I will try those next as soon as I can.