1000001101000/Debian_on_Buffalo #203: LSCHLv2: ata driver is hard resetting the SATA-Link on boot

cYrAx157, 2023-12-08:
hi there,
I have a linkstation live (ls-chlv2) and I installed debian bookworm using the debootstrap script.
When the device boots, I can hear that the harddisk is resetting a few times and after a few tries (sometimes it needs 20 or more retries) the boot succeeds.
In the dmesg, you can see that the ata driver is hard resetting the link.
The problem doesn´t occur on older debian versions (e.g. squeeze)
The HDD is a 4TB WD-Red "WD40EFRX".
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.55 (root@fbbf9fbb63a7) (arm-linux-gnueabi-gcc (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #12 Sun Oct  8 08:16:56 UTC 2023
[    0.000000] CPU: Feroceon 88FR131 [56251311] revision 1 (ARMv5TE), cr=0005397f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] OF: fdt: Machine model: Buffalo Linkstation LS-CHLv2
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=ttyS0,115200 bootshim=1
[    0.000000] Unknown kernel command line parameters "bootshim=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:on, heap free:off
[    0.000000] Memory: 48092K/65536K available (7710K kernel code, 1167K rwdata, 2552K rodata, 248K init, 235K bss, 17444K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: orion_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 11467562657 ns
[    0.000002] sched_clock: 32 bits at 167MHz, resolution 6ns, wraps every 12884901885ns
[    0.000065] Switching to timer-based delay loop, resolution 6ns
[    0.000421] Calibrating delay loop (skipped), value calculated using timer frequency.. 333.33 BogoMIPS (lpj=666666)
[    0.000467] CPU: Testing write buffer coherency: ok
[    0.000648] pid_max: default: 32768 minimum: 301
[    0.001489] LSM: Security Framework initializing
[    0.002109] landlock: Up and running.
[    0.002125] Yama: disabled by default; enable with sysctl kernel.yama.*
[    0.002538] AppArmor: AppArmor initialized
[    0.003150] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.003204] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.009762] cblist_init_generic: Setting adjustable number of callback queues.
[    0.009785] cblist_init_generic: Setting shift to 0 and lim to 1.
[    0.010327] Setting up static identity map for 0x8220 - 0x825c
[    0.010814] mvebu-soc-id: MVEBU SoC ID=0x6281, Rev=0x2
[    0.013532] devtmpfs: initialized
[    0.023427] VFP support v0.3: not present
[    0.023701] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.023768] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.024316] pinctrl core: initialized pinctrl subsystem
[    0.028222] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.029399] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.031050] audit: initializing netlink subsys (disabled)
[    0.033079] thermal_sys: Registered thermal governor 'fair_share'
[    0.033104] thermal_sys: Registered thermal governor 'bang_bang'
[    0.033118] thermal_sys: Registered thermal governor 'step_wise'
[    0.033129] thermal_sys: Registered thermal governor 'user_space'
[    0.033199] cpuidle: using governor ladder
[    0.033942] Feroceon L2: Enabling L2
[    0.033995] Feroceon L2: Cache support initialised.
[    0.037106] audit: type=2000 audit(0.028:1): state=initialized audit_enabled=0 res=1
[    0.049922] No ATAGs?
[    0.692202] raid6: int32x8  gen()    56 MB/s
[    0.760486] raid6: int32x4  gen()    56 MB/s
[    0.828303] raid6: int32x2  gen()    75 MB/s
[    0.896417] raid6: int32x1  gen()    57 MB/s
[    0.896433] raid6: using algorithm int32x2 gen() 75 MB/s
[    0.964134] raid6: .... xor() 45 MB/s, rmw enabled
[... 552 lines cut ...]
[  107.277156] ata1.00: cmd 60/18:78:70:1d:c4/00:00:00:00:00/40 tag 15 ncq dma 12288 in
                        res 40/00:78:70:1d:c4/00:00:00:00:00/40 Emask 0x10 (ATA bus error)
[  107.292924] ata1.00: status: { DRDY }
[  107.296646] ata1: hard resetting link
[  107.630340] ata1: SATA link down (SStatus 0 SControl F310)
[  107.710643] ata1: hard resetting link
[  111.003781] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl F310)
[  111.056973] ata1.00: configured for UDMA/33
[  111.061377] sd 0:0:0:0: [sda] tag#14 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=3s
[  111.070456] sd 0:0:0:0: [sda] tag#14 Sense Key : Illegal Request [current]
[  111.077472] sd 0:0:0:0: [sda] tag#14 Add. Sense: Unaligned write command
[  111.084217] sd 0:0:0:0: [sda] tag#14 CDB: Read(16) 88 00 00 00 00 00 00 c4 1d 48 00 00 00 08 00 00
[  111.093224] I/O error, dev sda, sector 12852552 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[  111.102385] sd 0:0:0:0: [sda] tag#15 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_OK cmd_age=3s
[  111.111455] sd 0:0:0:0: [sda] tag#15 Sense Key : Illegal Request [current]
[  111.118452] sd 0:0:0:0: [sda] tag#15 Add. Sense: Unaligned write command
[  111.125186] sd 0:0:0:0: [sda] tag#15 CDB: Read(16) 88 00 00 00 00 00 00 c4 1d 70 00 00 00 18 00 00
[  111.134162] I/O error, dev sda, sector 12852592 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 2
[  111.143247] ata1: EH complete
[  111.464565] ata1: SATA link down (SStatus 0 SControl F310)
[  114.743781] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl F310)
[  114.794786] ata1.00: configured for UDMA/33
[  172.147265] systemd-journald[218]: File /var/log/journal/c7330a3e93354933a2addc1a4d401f63/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.`

thanks in advance

1000001101000, 2023-12-08:
I have a theory, though I wouldn't have expected it to apply to this combination of device/drive.

Somewhere between Bullseye (5.10) and Bookworm (6.1) the EXT4 filesystem started enabling TRIM by default. This can cause problems for SATA controllers that don't support TRIM, the port expander on the TS-XEL is the main one I'm familiar with.

Normally you would only encounter this issue with SSDs, but it looks like some WD RED drives report TRIM support. I wouldn't think this device would have that problem since it doesn't have a port expander but I'm not certain it wouldn't.

To protect against this issue I added "nodiscard" to the mount parameters for the rootfs. It looks like you are using EXT4 for other partitions on the disk as well. Try adding "nodiscard" to the options for all EXT4 filesystems in your /etc/fstab and see if that helps.

cYrAx157, 2023-12-08:
good theory, but, no luck :-(
I tried a few things, and the only thing I can say is, that the last working distro is "stretch" with the 4.9 kernel.
Bullseye doesn´t work too, same issue.
Any other ideas ?
Maybe it´s time for the ls-chlv2 to rest in peace....

1000001101000, 2023-12-09:
Could you install `smartmontools` and then post the output of `smartctl -a /dev/sda`

cYrAx157, 2023-12-09:
here you go:

[smartctl.log](https://github.com/1000001101000/Debian_on_Buffalo/files/13624044/smartctl.log)

I don´t think it´s an hardware issue on the hdd side, but I don´t know if it´s on the Linkstation side.
Yesterday, I tried a different HDD (Samsung), same problem.
The only thing I can say is, i get a stable Debian if I use "stretch" (4.9).
Because of that, I really think, the problem is related in software.

Does it help if I install debian-stretch and post a dmesg or something ?

1000001101000, 2023-12-09:
I should probably start out by saying that I believe you when you say it’s likely not the hard drive. There are a bunch of similar issues out there for various sata controllers, a lot of those threads end in dismissive comments similar to “hard drives wear out bro”….. drives me crazy.

the SMART data helps me understand a little more about the drive. I might even have the same model here somewhere. It doesn’t look like the drive is reporting massive CRC errors/etc which could hint at some things.

I would have expected most of the kirkwood devices to have the same issues since I would have thought they all have the same sata controller …. but I would have expected to hear about this from a lot of people unless this is some how specific to this model or possibly types of drives.

Are you able to confirm whether the issue happens/happened on buster or bullseye? Narrowing down the kernel version that first had the issue might point us at what the issue might be.

You could also try EXT3, and possibly not mounting the data partition to see if there’s a filesystem component to the issue. The TRIM thing was specific to EXT4 on pretty recent kernels but there could be other such things that recently became default.

rems28, 2023-12-12:
Hello, for my LS-XHL with 256 MB amount of meory, the last chance to have a workable device is to keep the 4.19 kernel from buster (10) debian. With all new kernels, it's hangs like you.

cYrAx157, 2023-12-14:
@1000001101000 okay I will try a few things and report back.
@rems28 yeah, I can remember that I get a stable behaviour with a 4.x kernel so I will try buster first

rems28, 2023-12-16:
I compile new kernel 4.19.301 on this device with Debian .config from buster. After reboot, it continue to work like a charm and an uname command said me that Iam now on the new kernel.

1000001101000, 2023-12-16:
compiled on the device? I bet that took a while! That’s a good step forward.

ideally you could now try 5.10 and confirm that is broken…. then start trying kernels in-between to narrow down what kernel the issue started with. Once you’ve narrowed it down sufficiently we can look at the changes to relevant sata/fs stuff and try to determine what caused the problem.

you can probably save a lot of time grabbing armel “marvel” kernel packages from debian’s archive instead of building each one.

https://snapshot.debian.org/

1000001101000, 2023-12-16:
That patch is in a device tree for a different device, it wouldn’t have any effect on yours.

If you wanted to determine if the problem was with my kernel or Debian’s specifically you’d need to build the same kernel version to compare.

rems28, 2023-12-19:
I've tried with 6.1 kernel today and it hangs at boot.

1000001101000, 2023-12-20:
Excellent, it sounds like you’ve now confirmed 4.19 works and 6.1 doesn’t. If bullseye didn't work 5.10 probably doesn’t but i’d check that next. From that point there are relatively few versions to check between 4.19 and 5.10.

if you can narrow it down to that point we might be able to figure out what changes to the sata driver, filesystems, etc might be the cause and start working on a fix

pjt-15e, 2024-06-11:
I think I might have found the cause, though it took me a few days of "research".

As above my hard drive would hard reset several times during the boot process.  Examining dmesg output showed an error involving MPP pin 10 being assigned to power-hdd when already assigned to serial 0 (aka UART0).  Or words to that effect.  The dmesg log also showed that the hard drive was often being connected at lower than expected speeds.

After checking through the kirkwood-88f6281 hardware documentation, kirkwood.dtsi, and kirkwood-6281.dtsi, I edited the Bookworm device tree kirkwood-lschlv2.dts file, changing "serial@12000" to "serial@12100" so that UART1 using MPP pins 13 and 14 would be active, instead of UART0 which previously used MPP pins 10 and 11.  (Though serial output might be useful to help debug what's going on the connection points on the PCB aren't known to me.)

After making this change, generating and installing a new "debian_bookworm_armel.img" the boot process no longer has the hard drive resetting, and dmesg output no longer contains any warnings.  The NAS reliably boots from cold in about 65 seconds.  The dmesg logs also show the hard drive consistently using UDMA/133.

The above is probably not the best way to resolve the problem, but I'm happy with it so far.  I've learnt a lot about the Linux boot process, device trees, and Marvel Kirkwood processors, which has kept me entertained.

Anyway, thanks for Debian_on_Buffalo.

(I would do a pull request, but don't really know how!)

1000001101000, 2024-06-12:
Sounds like solid work to me. 

Might explain why I’ve not seen that with mine since I typically test with really low power ssds.

I’ll see if I can repeat your findings.

1000001101000, 2024-06-12:
I was able to confirm making that change for the UART made the error about MPP10 go away. I went ahead and updated the repo version right away.

the new dtb can be insalled by 

- copy it to `/etc/flash-kernel/dtbs/`
- run `flash-kernel` to generate new boot files
- `reboot`

I haven't verified the serial console works, trying that will be a task for another day.
https://web.archive.org/web/20160829014742/http://buffalo.nas-central.org/wiki/Serial_and_JTAG_port_LS-XHL

rems28, 2024-06-14:
Hello, do you thing that is suitable for ls-xhl too ?

1000001101000, 2024-06-14:
Almost certainly. 

Could you verify if you're getting that same MPP10 message in dmesg?

rems28, 2024-06-14:
I do not have cable for serial debug, but on a blank hard drive, I've tested the changes from the ls-xhl.dts file from the kernel.org source and the drive works like a charm on bullseye now. Before it was impossible to use that debian version after upgrade and wad impossible to use from scratch with debian installer. So I think that pjt-15e have certainly found the solution of the issue.

cYrAx157, 2024-06-14:
> I think I might have found the cause, though it took me a few days of "research".
>
> As above my hard drive would hard reset several times during the boot process. Examining dmesg output showed an [...]

wow, nice find !! It works like a charm on my lschl-v2 too ! Thanks for that. That patch should be pushed to debian´s repo too !

rems28, 2024-06-15:
Update to Debian 12 is good. Do not see any problem at the moment.

1000001101000, 2024-06-15:
I’ve updated the ls-xhl dtb with the same change and generated new installer images.