1000001101000/Debian_on_Buffalo #205: No network connection after successfull install Terastation TS-XL

andreasnsa, 2023-12-20:
Hi,
i have problems after installing debian on a terastation TS-XL. If installation is successfull and the terastation reboots (debian bookworm and also tested bullseye) there is no network connection to the terastation possible anymore. In installation mode its no problem, the terastation gets ip from dhcp. After reboot, display is green and version of debian shows in the display. I have tested both network ports, nothing. 
Did i missed something?

1000001101000, 2023-12-20:
The dhcp client the installer uses vs the one in the Debian install are a little different about the client id they send. I would guess that the terastation is getting a different ip address.

andreasnsa, 2023-12-20:
no, i see no dhcp leases in my dhcp server. only the old lease from installer, but the old ip is not reachable. tomorrow i'm getting serial cable, the one i have here (uart usb) only gives me hyroglyphs in terminal session (alle bitrates tested).
I hope i can then see something whats going on.

1000001101000, 2023-12-20:
serial connection should let you see what the device thinks is going on with the network.... at least I think there should be output for the link going up and for attempts to get a dhcp lease etc. 

Garbage in the serial output would imply it's connected but may have wrong settings, speed/parity/stop bits etc... could also be a bad connection I suppose. I have the details of the serial parameters on the wiki:
https://buffalonas.miraheze.org/wiki/Terastation_TS-XL#Device_Notes

I suppose you could also hook it up to a PC and verify if it's sending DHCP requests with a packet sniffer.... I haven't done much of that kind of work..... well, not with DHCP stuff

1000001101000, 2023-12-21:
I went ahead and tried an install with my TS-XL, it picked up an IP address after the reboot without issue.

closed 2023-12-26 by andreasnsa