1000001101000/Debian_on_Buffalo #212: Failed to create a file system

TimoEck, 2024-03-18:
Hey, 

I'm a beginner. It's my first try to change my Linkstation System to Nextcloud. 

The first step is to install ubunto. I followed https://www.youtube.com/watch?v=dfMHlLmsSF8. By Side I checked the wiki (https://github.com/1000001101000/Debian_on_Buffalo/wiki) and followed the way with **Option 1: Load onto Device Running the Stock Firmware**. 

Now I tried to install the ubunto. Followed the partion advices in the video and set the partions this way:
[image]
[image]
[image]

My next Screen and choosed option:
[image]

Next Screen:
[image]

And after a while (<1min) Don't be scared my cmd at windows doesn't show right at this install process:
[image]

If your need additonal Information about my System or my installation way please let me know. 

Additional:
I want to archive Raid1. I don't know how to archive this and can I do it now with the partion or later?

1000001101000, 2024-03-18:
sda6 and sdb6 are components of RAID#10. if you change those to unused it should be able to move forward.

you can specify a mountpoint for RAID#10 if you wish or you can wait till after you get into debian

TimoEck, 2024-03-18:
Hello, thank you so much for your fast responde! 

Sorry for my Questions. But I got the RAID#10 "use as: do not use"
[image]

If I change RAID#10 to ext4 and the mountpoint to: "/home", will it save all files there 
[image]

or should I "use as: do not use" for both 2TB Hard Drives?

TimoEck, 2024-03-18:
I changed RAID#10 to ext4 and tried to set the mountpoint to /mnt/data. 
Changed the filesystem sucessfully, but gave me the same error. "The ext4 file system creation in partition #6 of SCSI1 (0,0,0) (sda) failed.

TimoEck, 2024-03-18:
Worked. Debian is installed. Thank you very much for your help!

closed 2024-03-18 by TimoEck