1000001101000/Debian_on_Buffalo #218: Successful install on LS-420, but slow performance

cgmorlock, 2024-06-05:
Installation was successful, and I was able to keep and use all existing partitions, including RAID-1 array, however getting periodic issues where it will 'pause' for 30-40 seconds.

dmesg  shows following continuously repeating:

status: { DRDY }
hard resetting link
SATA link up 1.5 Gbps (SStatus 113 SControl F310)
configured for UDMA/33
retrying FLUSH 0xea Emask 0x4
EH complete
exception Emask 0x0 SAct 0x0 SErr 0x0 action 0x6 frozen
failed command: FLUSH CACHE EXT
cmd ea/00:00:00:00:00/00:00:00:00:00/a0 tag 19
res 40/00:01:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)

Any suggestions on how to remedy?

1000001101000, 2024-06-05:
I’d start by looking at the SMART data for relocated sectors and errors, I’d also run a “long” SMART self test.

cgmorlock, 2024-06-08:
I was able to resolve the issue by rebuilding the array.  Not sure why I had to but when I specified the array and devices on the mdadm command rather than relying on the --scan, it rebuild the array and I am no longer seeing any issues.

mdadm --assemble  /dev/md10 /dev/sda /dev/sdb  took about 26 hours to complete, but that seems to have resolved the issue.

1000001101000, 2024-06-08:
I’d still check the SMART data on both drives.

the performance loss could be largely the array becoming degraded and rebuilding will fix that. The question is why it became degraded in the first place. The errors you posted indicated it’s getting errors reading a drive, that could mean the drive is starting to fail. If so it’ll probably become degraded again in the near future.