1000001101000/Debian_on_Buffalo #223: Restore .bak files to go back to stock on LS1200D

pcp13, 2025-02-05:
I have a LS1200 device with two 4Tb hard drives. I used the python server method to copy the appropriate new uImage.buffalo and initrd.buffalo files to the /boot folder of the device. I also backed the stock versions of the two files prior to that, also in the same /boot folder. 

I rebooted the device and am able to ssh into the device with the installer@<device ip> method. I am at the installer screen (hopefully the screenshot shows up here). 

[image]

Is there a way to go back to the /boot folder and revert the two .buffalo files to stock version? If I go into shell from the first installation menu, there is a message that the "hard disk file systems are mounted on "/target". However, I can't find the /target mount point at all. When I run ls /sys/block there is no output; mdadm is not found. 

I don't have an external adapter to read the drives outside of the device. I wonder if restoring stock in EM mode is the only remaining option. I am on a laptop with win10 and linux mint dual boot so emergency restore is possible. Thanks.

1000001101000, 2025-02-06:
if you rebooted into the installer again it likely means you didn't set up /boot. The easiest way to correct it is probably to just run through the installer again and specify the /boot mount point for the boot volume this time. 

alternately you can take the installer up the the partition screen (so it loads all the needed disk/etc drivers) then drop to a shell (or connect via another ssh session) and mount /boot and make whatever modifications you want from there.

pcp13, 2025-02-06:
Thanks for the reply. I had one typo in my original post. The device is a TS1200D not LS1200. 

I tried the second method you suggested. I was able to mount the old /boot partition and change back the two .bak files to .buffalo files. I tried rebooting the machine and now it appears to be in some state of error. Both the function and power LEDs on the front of device blink; the individual disks' LEDs are not indicating disk error so that's some good news. It doesn't show up on the network at all, not via ping, acp_commander or on my router's admin panel. 

Need to read more about how to recover from this state. There is no data on the disks so I might have to try some drastic method to get the device to boot normally.

1000001101000, 2025-02-07:
I would think it would boot to EM mode with those files back in place.

there's always this procedure:
https://buffalonas.miraheze.org/wiki/Restoring_Stock_Firmware_without_TFTP

pcp13, 2025-02-07:
I powered down the device by unplugging the power and rebooted it. This time, the device showed up on my network and acp_commander could see it. When I booted into Windows, the device showed up as in Emergency Mode in NAS Navigator. From there, it was fairly straight forward to restore the stock firmware.

It's back to stock and syncing the Raid1 array. It shows as over 2000 minutes remaining and these are empty drives so I am not sure what it's doing. I may need to look into stopping whatever it's doing right now. I don't want the empty device to be spinning for over 30 hours.

Thanks for you help!!

closed 2025-03-13 by pcp13